kerberos-io/agent #186: Something went wrong while verifying the camera settings: not enough bits

flyingspud, 2025-05-19:
I am trying to add RTSP feed from Hikvision cameras to Kerberos agent v3.1.8, but encountered the error message "not enough bits" when either adding the RTSP stream directly into my Docker compose, or via the agent GUI.

The following have been tested:
- RTSP streams have been tested in VLC, OBS, and Frigate to be OK
- Using both direct RTSP from camera, and from go2rtc restreaming generate the same error message
- streams using H265 encoding with no audio
- tried RTSP streaming from TP-Link and NETUS DVR, same error

Setup:
- Synology DS420+ and Docker
- 5x Kerberos agents (currently one spooled up for testing)
- Minio and Kerberos Vault setup OK

flyingspud, 2025-05-20:
@cedricve cameras were previously connected to OBS for direct stream and to Synology Security Center for CCTV function. They were subsequently consolidated under go2rtc since I needed additional RTSP streams and Hikvision is limited to 2 concurrent stream per camera.
Anyway, I have switched the encoding to H264 and got it working. Weird given that you mentioned H265 supposed to work but the agent is not biting even when I entered the direct RTSP stream details. Managed to get 4 out of 5 cameras to work, only one of them is still not working at the moment due to "i/o timeout error" from go2rtc and direct stream.

flyingspud, 2025-05-20:
@cedricve let me know if you need any form of debug log. As long as it does not contain any identifiable or sensitive information, I should be able to share it with you for further troubleshooting and aid with development works.

apetrichkovich, 2025-05-21:
Hello. I have the same issue - can't to create camera with RTSP stream.
`Something went wrong while verifying the camera settings.: not enough bits`

cedricve, 2025-05-22:
@apetrichkovich which camera are you using? could you share a bit more details?

apetrichkovich, 2025-05-22:
hello @cedricve 
WiFi IP camera QC013 CAMERA Y7 (sometimes it is selling as QC011).

[image]

flyingspud, 2025-05-23:
@apetrichkovich does your camera have the option to output the camera stream as H264? If yes, give that a try and see if go2rtc picks it up.

apetrichkovich, 2025-05-24:
Hello @flyingspud,

I’ve converted my RTSP stream to H.264 using MediaMTX and FFmpeg, and successfully added it to Kerberos Agent. I can view recorded clips, but the Live View either doesn't work consistently or freezes frequently.

From my experience so far, I assume the following:

1. Kerberos Agent does not support H.265 streams out of the box.
2. Live View functionality is unreliable or possibly requires additional configuration or components.

Could you confirm if this is expected behavior, and advise if there are recommended steps to improve Live View performance?

Thanks in advance!

flyingspud, 2025-05-30:
@apetrichkovich I personally encountered streaming issue with H265, but this is something the developer will need to relook since it is supposed to be supported out of box.
As for live view, I can't confirm as it is working fine for me, but I am using go2rtc as my restreaming server. If this is a persistent issue for you, you may need to raise a separate thread.